GaloisInc/csaf #140: Pre-release checklist

podhrmic, 2021-11-02:
Remaining items:

- [ ] CI passing
- [ ] import remaining examples from old CSAF
- [ ] import remaining notebooks from old CSAF
- [ ] properly merge/rebase flight gear visualization from `develop`
- [ ] unify how to run new CSAF (conda, docker, pip, something else)
- [ ] update README
- [ ] update Changelog
- [ ] update claims about using foreign (non-python) components, maybe provide an example
- [ ] update/rewrite/remove Software Requirements Specification document

EthanJamesLew, 2021-11-10:
The branch `feature/csaf-examples` imports remaining examples and notebooks

cslockett, 2021-12-17:
Incorporated updates for Software requirements specification.  Action complete
Agreed not to merge and support flight gear for final client deliverable.  ready to close this issue

closed 2022-01-18 by EthanJamesLew